TomBric/stratux-radar-display #5: Bluetooth issue sound

mofaheiti, 2021-01-30:
Hello Tombric,

I Have the setting you Explain:
- Raspberry Pi zero WH
- 3 inch waveshare eink display
I tried 2 ways: the complete image; and the expert image installed from rasperry pi image till your code.
both installation ways do run fine, but i have a bluetooth problem.
i can connect the bluetooth headset you explain (no problem) but i have no sound output.
The headset shows connected, the raspberry too. but no sound. ( i tried 3 different bluetooth headsets)
when i try this command :  
aplay /usr/share/sounds/alsa/Front_Center.wav 
i hear the testsound :). So by this way it works, but still no sound by the radar :(.

Is it possible, that the sound goes to the hdmi or other port?
Or do i have to update blueaz?

Kind Regards Michael

TomBric, 2021-01-30:
Hi Michael,

did you start the radar with the option "-s"?
Does the radar show a Bluetooth symbol in the right down corner?

Thomas

mofaheiti, 2021-01-30:
Hi TomBric,

Yes, i start with "-s" (in your image it is automatically written)
Yes the Display shows the Bluetooth icon in the right down corner when connected. ( i tried the oled version too, and the same problem)
The display works perfectly! Thanks for your work.

if you google "bluetoothctl no sound" there is the issue, but i tried 20 things and nothing worked.
The only thing i have different to your description, that my headset do not have a secure code (is that the point?)

but i dont understand: the command "aplay /usr/share/sounds/alsa/Front_Center.wav" works, i hear the sound. so the bluetooth connection is stable.

if you have an idea, i can test it

TomBric, 2021-01-30:
Hi Michael,

Hmm. Can you log in to the raspberry as pi and try

espeak-ng "Hello Test"

Thomas

mofaheiti, 2021-01-30:
:) that works

mofaheiti, 2021-01-30:
But only this text, the radar which is parallel on, says nothing

TomBric, 2021-01-30:
That's a good sign. At least bluetooth works, which is tricky on the raspberry.

When you start the radar manually, can you check whether there is an output like?
"INFO: espeak-ng not initialized"

To start manually:
- log on as pi user
- stop the running radar process by:
      ps -uax | grep radar
      kill xxxx      -> xxx = the number in the second column where radar.py is referenced
      recheck   via ps -uax | grep radar
- then start radar manually by:
      cd /home/pi/stratux-radar-display/main && python3 radar.py -s -d Oled_1in5 -c 192.168.10.1

Then please check the first lines, whether there is a log on "espeak-ng" or "SPEAK".

Thomas

mofaheiti, 2021-01-31:
Good morning Thomas,

Ive tried the points
when i start it after kill manually i got sound :) without doing anything else.
but i see only this 3 lines (no more, so i cant check if speak or espeak-ng):
-2021-01-29 18:30:59,713 > Measured Display Refresh Time: 1.192408561706543 seconds
-TrafficHandler waiting for ws://192.168.10.1/radar
-SituationHandler waiting for ws://192.168.10.1/situation

seems that one service starts to early.
i tried this:
File: /home/pi/stratux-radar-display/image/stratux_radar.sh
sleep 100
cd /home/pi/stratux-radar-display/main && python3 radar.py -s -d Epaper_3in7 -c 192.168.178.10

The Display starts later, but no sound

I hope this is comprehensible

Michael

TomBric, 2021-01-31:
Hi Michael,

can you remove the "&" in the last line of  stratux_radar.sh and try again?

Thomas

mofaheiti, 2021-01-31:
Hi,
I tried without the "&"
The same, no sound

When manuel kill and start i have the sound
Then restart, no sound

TomBric, 2021-01-31:
Hi Michael,
at least this works, that is very good. 
I can not reproduce it now. I will produce a new release the next days which will give us easier debugging infos. I will inform you here.

Thomas

TomBric, 2021-02-02:
Hi Michael,
I could reproduce the "no sound" error today. I will check the next days how to fix it.

Thomas

TomBric, 2021-02-05:
Hi Michael,
I found the reason for this issue: pulseaudio.
system.d only spawns pulseaudio if there is a login session.
 
So the solution is to generally start pulseaudio as a system daemon. This needs several steps in configuring the operating system. I will put this steps into the configuration skript. If you like you can already test it interactively:

> sudo sed -i '$ a load-module module-bluetooth-discover' /etc/pulse/system.pa
sudo sed -i '$ a load-module module-bluetooth-policy' /etc/pulse/system.pa
sudo addgroup pulse bluetooth
sudo cp pulseaudio.service /etc/systemd/system/
systemctl --system enable pulseaudio.service
systemctl --system start pulseaudio.service

/etc/systemd/system/pulseaudio.service needs to have the following content (will be provided in the repo):

> [Unit]
Description=PulseAudio system server
[Service]
Type=notify
ExecStart=/usr/bin/pulseaudio --daemonize=no --system --realtime --log-target=syslog
[Install]
WantedBy=multi-user.target

I will provide everything in the configuration skripts and also in the image.
Audio and bluetooth is really a lot of hassle on the pi.

Thomas

mofaheiti, 2021-02-06:
Hey Thomas,

yes it works, thank you for your work.
but only one thing. when i restart the device, the bluetooth do not connect automatic.
I have to do

bluetoothctl
remove <device>

and then reconnect via your instruction

mhh, very tricky this thing,
or is ist easyer to flash it on a raspberry pi 2 or 3 and do the audio output over the audio jack

Michael

TomBric, 2021-02-06:
Hi Michael,
good that is working now.
For automatic reconnect do the following: 

> sudo sed -i '$ a load-module module-switch-on-connect' /etc/pulse/system.pa

I will also add that in the configuration skript.

Thanks
Thomas

mofaheiti, 2021-02-06:
Hi Thomas,

sorry but this do not work.

I tried this after a boot manual:
-bluetoothctl
-connect <deviceid>
-exit
and then it works perfekt. but only for this season. By reboot i have to connect again manual

so i tried this solution:
-sudo nano /etc/rc.local
-echo "connect <deviceid> \nquit" | bluetoothctl
but it doesnt work

then i tried
sudo nano /home/pi/stratux-radar-display/image/stratux_radar.sh
and add in a new line
-sleep10
-bluetoothctl
-sleep10
-echo connect "<deviceid>" | bluetoothctl
-exit

but that doesnt work too.
I don't know how to fix.

Michael

TomBric, 2021-02-07:
Hi Michael,
I recked that. My headset does a reconnect, when switched on after a reboot.
Maybe it is because you have inconsistent image now.
Can you flash a new image and try again?

Thomas

mofaheiti, 2021-02-09:
Good evening Thomas,

so I've tried a few things and found a solution for my setup.
my goal is to print a case and build the display hard wired in my plane.
when the case is ready, i will link it here.

My setup:
raspberry pi zero wh
waveshare 3.7 eink hat
bluetooth bk8000 hard wired on 5V pin raspberry (audio to the intercom)

The installation:
- expert setup with your instruction (works fine!)
- bluetooth pairing with your instruction (works fine!)
- connect by reboot doesnt worked, so i tried many things.
=> solution: 
- sudo crontab -e
add this line and save
- @reboot sleep 70 && /bin/echo -e 'connect XX:XX:XX:XX:XX:XX \n quit \n' | bluetoothctl

Thank you for your great work!!!!!! :)

TomBric, 2021-02-09:
Hi Michael,
good that you found a way for your setup!  
If you want hardwired sound to your intercom, using a Pi3B would also be a solution. Of course this requires some software modification, but it's certainly possible. So check your setup and if it does not work reliable, I can implement some tweaks for onboard sound. Using the Raspi 3B also makes the epaper display a little bit quicker.

Thomas

closed 2021-02-09 by TomBric