TomBric/stratux-radar-display #12: install script failing on stratux eu023 + PI 4

gpisapia, 2021-03-02:
Hello, 
I'm trying to use your nice radar module on latest stratux europe. Actually made it straightforward until:
-Pulse audio wasn't installed by default, had to do it.
-Your command line include -s assuming BT is on, installed and working correctly. I suggest to spend a little more explanantion on this, although it is not a priority item to make it work.
-By default, SPI in not ON when standard stratux image is burnt on SD card, you have to pass by raspi-config to enable it.

Otherwise I could get it working.! Great!

TomBric, 2021-03-04:
Hi gpisapia,
good to hear that it is working.
Indeed the configuration script is only built for a separate installation on a Pi Zero, based on the standard raspbian os.

Did you get Bluetooth working on the Stratux image? BT is the main reason why I am using standard raspbian (not lite). Maybe if the raspi foundation gets this fixed together with pulseaudio in the lite version, I will also move there and build a configuration skript on the Stratux...

gpisapia, 2021-03-04:
Actually BT is not working well on stratux 023 because it's probably based on a buster lite image whose BT was not correct. My PI4 run BT correctly on the latest buster but I am struggling to have it working with st023. 
So far stratux-radar-display works well but without the sound.

closed 2021-03-10 by TomBric