TomBric/stratux-radar-display #29: Not possible to flash the file

Dalotel, 2021-04-20:
Hello Sir,
When I try to flash the file with Etcher I have this message.
I have a very low level knowledge about Raspberry things.
Could you help me?
Thanks
Armel
[image]

TomBric, 2021-04-21:
Good Morning Armel,
I just rechecked both images. Both worked for me. The filename of the file in your error prompt is strange. It should be stratux-radar-display-epaper-v001h.zip or stratux-display-oled-v001h.zip. No idea what went wrong.
I propose to use "Rasperry Pi Imager" instead of Etcher. Has more options. Just start it and then select "Operating System -> Own image". 

Thomas

Dalotel, 2021-04-21:
Thanks for your answer Thomas. With imager everything works.
Finally, my help request go further than that. As I said I am a totally beginner in the Rasp World.
I assemble a Stratux device with a PI4B, two Nano Three SDR, TTGO T beam. I follow closely the procedure and everything works :-).
I don't like to use my phone as display as I don't trust on it. So it's why, when I discovered your work I was really happy.
At the beginning, my aim was to connect the Stratux to my Garmin AERA 660. But it seems that garmin make it impossible as it is not using GDL90 protocol.
I order a 1.5" OLED screen and waiting for it arrival. I also have a 2.2 TFT SPI 240*320 bought several month ago.
I am interested to run the display directly on my Stratux device or connect by wires but I am totally lost how to do it.
I don't understand where, when to install your program and the Stratux one.
I don't have AHRS installed, as I have everything I need on my plane.
I make things short I need something almost ready to go if it's possible!

I understand this is time consuming so I will understand you decline.

Thanks a lot

Armel

TomBric, 2021-04-21:
Hi Armel,

good to hear that it works. 
I have a small section in the readme how to install my software directly on stratux.
But there is also a complete image available of VirusPilot with both parts, see here:
https://github.com/VirusPilot/stratux-pi4/tree/v2.1

The problem of operating the display directly on stratux is, that you do not have Bluetooth. I think it is a very important feature to have voice warnings in the headsets (e.g. with Bosa A20). That's why I recommend an additional Raspi Zero which is not very expensive.

Thomas

closed 2021-04-23 by TomBric