TomBric/stratux-radar-display #39: No connexion

Dalotel, 2021-10-08:
Bonsoir,
After some weeks of inactivity, I restart my stratux project. I update my PI4 by the lastest stratux edition (Stratux Europe Edition 1.6r1-eu026.)
I update my OLED/zero with your last file but it refused to connect with my stratux.
During startup, it show it will connect to 192.168.10.1
On my phone the Stratux IP address is 192.168.10.1.
I checked in the file mention on standard setup and it's the right IP address.
Could you help me.
Thanks
Armel

TomBric, 2021-10-09:
Hi Armel,
I just checked with two fresh images:
- eu026
- oled 1.1
The display connected to the stratux eu026.

Did you change any wifi settings on the stratux (or on the display) ?

Thomas

Dalotel, 2021-10-09:
Bonjour Thomas,

Thanks for your answer.
No, I didn't change any setting on Stratux.
I didn't install the buttons because I don't have baro sensor.
I am going to reinstall everything from the  beginning, even redownload
software and I will see.
I'll keep you update

Thanks

Armel

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Dalotel, 2021-10-14:
> Bonsoir Thomas,
>
> As I said, I reinstalled everything. I still have the same message.
> I send you some photos.
>
> I hope you will find the problem. I didn't see the zero on my WiFi phone
> settings. Is it normal?
>
> I am waiting your advice before to buy another pi zero.
>
> Best regards
> Armel

Bonsoir Thomas,

As I said, I reinstalled everything. I still have the same message.
I send you some photos.

I hope you will find the problem. I didn't see the zero on my WiFi phone
settings. Is it normal?

I am waiting your advice before to buy another pi zero.

Best regards
Armel

TomBric, 2021-11-09:
Hi Armel,
good to hear! 
On a Pi4 you could get the display also working, but you should do manual "expert" install. The images are for Pi Zero only.

Thomas

closed 2021-11-09 by TomBric