TomBric/stratux-radar-display #47: E paper display frozen on stratux eu027

mofaheiti, 2022-01-11:
Good evening Thomas,

I have an issue.
First, thanks for your great work! 
My Hardware:
- RPi 3B
- 2 SDR on usb (1x1090mhz 1x868mhz)
- 1 T-beam with bme280 and gps on usb
- experimenteal usb soundcard (works fine ;) )
- connect the e-paper display to the pins you explain

My Setup:
- format the sd card -> complete overwrite with sd card formatter
- Flash the card with balea etcher -> new eu 027 version
- Installing with putty ssh your stratux radar software (the steps you descripes)

Now the issue:
- stratux starts normal
- the display starts normal
- when no gps antenna is connected, the display works the hole time and shows "no gps". the right dot is moving.
- when i connect the gps antenna, the display works and "no gps" has gone. 
- now when no traffic is in the air, the display works.
- if the first traffic will receive, the display is freezing. (USB sound cards keep working an i hear the next traffics). But the display shows nothing and the right moving point is freezing too.
do you have an idea?

Michael

TomBric, 2022-01-11:
Hi Michael,

there are problems with one of the imported libraries. Pillow 9.0, used for graphic drawings,  produces an illegal instruction. I am working on it by importing an older pillow version during the build. New images will be released soon, latest tomorrow.

Thomas

TomBric, 2022-01-11:
Hi Michael,
can you do me a favor and test the following:
- login as user pi on the stratux
- execute command:  pip3 install Pillow==8.4
- reboot

Unfortunately I do not have an epaper display directly connected to my stratux currently. So give me a short feedback whether this is now working. On the Pi Zero it did the job ...

Thomas

mofaheiti, 2022-01-12:
Hi Thomas,

first i tried this:
- login as user pi on the stratux
- execute command: pip3 install Pillow==8.4
- reboot
-> the display still freezes .

second i tried this:
- login as user pi on the stratux
- execute command: pip uninstall pillow
- reboot
- execute command: pip3 install Pillow==8.4
- reboot
-> the display still freezes .

third i tried
- new flash stratux
- change configure_radar_on_stratux.sh ->sudo TMPDIR=/home/pi/tmp pip3 install Pillow==8.4
-> the display still freezes .

nothing works, everytime the same problem.
do you have an another idea? i can try and test ;)

Michael

TomBric, 2022-01-12:
Hi Michael,
thanks! I think I have to debug myself. I will setup an epaper display in the same setup like you tomorrow and check.
Thomas

TomBric, 2022-01-13:
Hi Michael,
I could reproduce the error from time to time. It seems to be a performance issue. When lot of events or messages are received, the SPI interface, which speaks to the display is no more working. 

Can you do the following and retest?
- in stratux-radar-display/main/displays/Epaper_3in7/epdconfig.py:
      change:   self.SPI.max_speed_hz = 32000000    
      to:           self.SPI.max_speed_hz =   8000000               (8 Mio means 8 Mhz)

You can find this definition in "def module_init". Changing can be done with some editor e.g. nano.

If this helps, I will put it into the general code. It makes the display interface a little bit slower, but its only 0.02 secs slower.

Thomas

TomBric, 2022-01-14:
Hi Michael,

I did some more tests yesterday evening: It's the OGN-module that causes the freezing. If you disable 868 everything should be fine. When the ogn-rx-eu process starts, reading of the gpio pins is disabled and the "busy" indicator of the display is always set. I will check together with the stratux developers how to mitigate that. ogn should not block the gpio seriell input.

Thomas

mofaheiti, 2022-01-14:
Hi Thomas,
Ok i did the test with your max_speed yesterday too. It doesn't work.
I will try this weekend disable the 868 mhz and test again.
Thanks for your trouble, I hope you and b3nn0 find an easy solution

do you have the same problem with the oled display?

Michael

TomBric, 2022-01-14:
Hi Michael,
no, I did not yet experience that. Oled has no input via GPIO (no busy flag), it just writes to GPIO.

Thomas

mofaheiti, 2022-01-14:
Hi Thomas,
Ah ok, i understand.
Now i testet switch 868 "off". Everything works fine.
another question: Can the switch "Persistent Logging" be turned off again after installation?

Michael

TomBric, 2022-01-15:
Hi Michael,
good to hear, its working!
No, persistent logging needs to be on. 
If you put it off, everything that you installed will be lost. There will be no data left, since the partition is no more on the sd card. But dont worry, this is how Stratux  worked for long times without corrupting the sd. And shutting down never really worked in stratux till we changed it several weeks ago.

Thomas

TomBric, 2022-01-24:
Hi Michael,
I got no response from the ogn developer Pawel till now. I checked again, it's the ogn-rw that also uses the SPI interface on the Pi and creates conflicts. Currently there is no option for ogn-rw to switch this behaviour off. So for now, nothing I can do. If you want an Epaper-display you have to invest in an additional Pi Zero ...
Thomas

closed 2022-01-24 by TomBric

mofaheiti, 2022-01-24:
Hi Thomas,
ok, thanks for your support!
so i will build my last setup with the zero.
Michael

TomBric, 2022-01-27:
Hi Michael,
I am in contact with the developer of ogn-rx-eu and there is now a ogn version available that does not use the GPIO pins. I tested that and everything works. I will check with b3nn0 to have that included in the upcoming stratux versions. 
In any case: using an additional zero is safe, no conflicts and bluetooth will work as well.
Thomas

mofaheiti, 2022-02-01:
Hi Thomas,
Ok thats sounds good.
I will wait till b3nn0 realease the next version.
Than i test again ;)
Michael

gbraux, 2026-10-05:
Hello,

FYI, I was hitting this bug, and I do confirm that Stratux 2.0 Pre-5 includes the fix, no more frozen screen when OGN is enabled !
Note that Pre-4 (the current "stable" release) is not including the fix.

Guillaume.