TomBric/stratux-radar-display #52: E-Ink Screen freezes

cavok2, 2022-04-30:
Hi Tom
 Love the project, and I have it up and more-or-less running, but only with the 1090 protocol. Once I enable the 868 dongle in the Stratux settings, the 3.7" E-ink Waveshare screen freezes. The Stratux continues to work since I can still receive indications via web interface/EFB, but I can only get the screen to work again if I disable the 868 protocol and reboot with only the 1090. I thought that maybe it was an energy issue (running on a Pi Zero 2W with USB HAT (connected via wires from USB board to display), or on a Pi 3B), but I have the same negative result regardless of which energy source (power bank with 3A output, or official Pi power source).
I am using the newest version of Stratux. Any help would be greatly appreciated

TomBric, 2022-04-30:
Hi cavok2,

hmm, I will try to reproduce it to be able to understand, what brings the epaper to freeze.
Can you please indicate:
- what stratux version are you using?  Please give me the exact version like written in the web-status window. e.g. 1.6r1-eu027
- what epaper version are you using? E.g. 1.61, as indicated during startup of the screen.
- what flarm device are you using?
- what GPS device are you using (see status, e.g. "SoftRF dongle NMEA protocol")
- what happens when the screen freezes? No aircraft are displayed any more? What about the "keep-alive" indicator on the right, is it still changing?
In case you are an "expert" user, please login as pi via ssh on the Zero2 of the display. Stop the running "radar" process, change directory to "stratux-radar-display/main" and run "python3 radar.py -d Epaper_3in7 -v 3".  This will give a verbose output and should indicate why the screen freezes.

Thomas

cavok2, 2022-05-01:
Hi Thomas,

Thanks for the quick response :-)

- Stratux version v1.61r1-eu027 (3d07b461d7)
- Radar 1.61
- GPS hardware: USB u-blox 7 GNSS receiver (NMEA protocol)

- "what happens when the screen freezes?":

> _keep alive_ stops moving

> aircraft displayed, but no update/do not move

> sorry, not an expert user

Testing indoors shows that if there is not a GPS fix, then the screen does not freeze, however, I am not getting an OGN signal either.

TomBric, 2022-05-01:
Hi cavok,
I just did a test and used exactly the same hardware (Nano2) + ublox 7. No problems on my side.
If there is no GPS fix, the OGN receiver software does not start. Strange is that the Zero completely stops, when the OGN receiver starts running. 

Just to be sure, that I understood your setup correctly: You connected the Epaper screen to a Pi Zero? Stratux itself is running on a different Pi 3B hardware?

(Running display and stratux on only one Raspberry Pi shows this effect currently: The OGN receiver blocks many IOPins and so the display is no longer responding. The way around currently is a separate Pi Zero for the display. But this will be solved soon. Together with the OGN guys a new version of the OGN software is already included in the new Stratux version, but there is not yet a new release for flashing available. Anyhow, i recommend a separate Pi Zero2 for the display, it just takes a few bucks)

Thomas

cavok2, 2022-05-01:
Hi Thomas,

I built my current set-up with a Pi Zero 2W and I am only using a Pi Zero 2W in my current Stratux set-up. I was pretty sure the same problem happened with the Pi 3B, but I'll check later again today. 
I do have a Pi Zero W laying around. How would I set it up so the 2W does the Stratux "heavy lifting" and the Pi W the display, if that is possible? 
Or should I use the GPIO pins for the display off of the Zero 2W and attach directly to the display?? I can make a work-around for the buttons.

Thanks again for your help!

TomBric, 2022-05-01:
Hi Cavok,
yes, if you are using stratux and display on the same hardware you currently get this problem. As said, upcoming version of stratux will solve that. 

So best and standard setup is:
- Zero 2W for stratux without display
- an additional Zero W (or Zero2)  for the display only including the buttons

For the Display-Zero, just use the image provided on github.

Thomas

cavok2, 2022-05-01:
Hi Thomas,

just tried out my RPi 3 with the same bad result that I had with my Z2W. No panic and I‘ll wait for the update. Thanks for your effort!

closed 2022-05-01 by cavok2

gbraux, 2026-10-05:
Hello,

FYI, I was hitting this bug, and I do confirm that **Stratux 2.0 Pre-5 includes the fix**, no more frozen screen when OGN is enabled !
Note that Pre-4 (the current "stable" release) is not including the fix.