TomBric/stratux-radar-display #53: Networking headless

christianhbeck, 2022-05-08:
Hello,

i am looking to get the Bluetooth audio feature to my Stratux. So I flashed the provided image to a Raspi zero w. The zero w does not have one of the supported displays attached. 

The pi zero does not connect to my Stratux, which is a current Europe edition with the default networking settings in place (ap and client).

i have attached a keyboard and hdmi Display to the Raspi zero for debugging. Radar.py is Running. Ifconfig showes no wifi Connection.

Is the display required?

regards,
Christian

TomBric, 2022-05-08:
Hi Christian,
I never tested the display-software without an display. It could work, if you select the Oled_1in5 version, since this has no feedback to the zero.
Nevertheless the wifi connection should work in any case and  the display should connect to the stratux network. 

What is your wish concerning audio feature? Do you want to have the bluetooth warnings without an display? How do you control your display and stratux? I just want to understand your setup and maybe I can test and modify the display software accordingy ....

Thomas

christianhbeck, 2022-05-08:
Hi Thomas,

i will try the oled image. Can you give some recommendations how to check the networking step by step?

I am currently investigating what Setup is ideal for me. My current thinking is that I want to add a small display to the panel. It should show map and traffic (I fly track up…so radar is the same orientation for me as map). Todays smart phones are to big for ad hoc integration into the panels of the ac I am flying and I do not want this display on my lap. Toward this end I have selected a 4 inch 720x720 square Touch  display. This is driven by the pi3b I have in my stratux and connected via cable. On the Stratux I have installed xcsoar and enroute. I have both working. I am undecided which one to stick with. Enroute is Limited by the hw, but with some tweaks (running a single x session without windows manager) I can at least avoid swapping. I like it better than xcsoar for its good visuals. Xcsoar on the other hand has elevation data and can display it well, performs better on low power hw,…. 
This setup should be there for low frequency visual scan, in case there is doubt about the situation.
For traffic I want to look outside and get augmentation via audio. I wanted to try the Bluetooth Audio route even if it means more hw since it integrates well with my headset. I want to have this to cover better the phases of flight when not being controlled.

regards 
Christian

TomBric, 2022-05-08:
Hi Christian,
I am not a fan of these LCD displays for use in our aircraft. Move it into the sun and check whether you are still able to read it. That's the reason why I highly recommend the Epaper displays.

Concerning your networking issues:  "AP+client" mode often raises problems due to driver issues on the raspi. I would recommend to switch to "Access Point" only mode with the Stratux. Include your PC as well into the stratux wifi and then try via ssh. 

Thomas

christianhbeck, 2022-05-08:
Hi Thomas,

i agree on the readability of e paper. I struggle to find one with sufficient number of colors (ideally 5+), good size (3-6 inch) and touchscreen. If you are aware of such a display I would be delighted.

The ap only mode fixed my networking issue. 

Thanks
Christian

closed 2022-05-08 by christianhbeck