TomBric/stratux-radar-display #54: USB soundcard problem

OdomLinx, 2022-06-03:
Hi,

i've a new usb soundcard (the other one dies) and no sound.

i found the problem in radarbluez.py i had to change MIXERNAME, my usb soundcard mixers name is not SPEAKER

Snip:
```
> Audio: Available Card:bcm2835 Headphones Mixer: Headphone
> Audio: Available Card:TX42BD100 Mixer: Headphone
> Audio: Available Card:TX42BD100 Mixer: Extension Unit
> Radarbluez: Mixer not found!
> SoundInit: Bluetooth active:False ExtSound active: False ExtSound volume: 50.
```
after change MIXERNAME to "Extension Unit" sound is working.

I dont't try headphone, while the name is double.

TomBric, 2022-06-05:
Hi OdomLinx,

yes, your approach was the right one. Depending on the USB hardware it may happen that other mixers are supplied by the cards.
To solve it also for future releases I will add an option.
So in future the option '--mixer "Extension Unit" '  in stratux_radar.sh will do the job. As soon as I have some time I will include it in the new version 1.8.

Thomas

TomBric, 2022-12-23:
Hi OdomLinx,
as promised this feature is included in version 1.8. You can specify "-mx MIXERNAME" as an option in your stratux_radar.sh script.
Thomas

closed 2022-12-23 by TomBric