TomBric/stratux-radar-display #55: epaper 2in7b

rudiratlos, 2022-07-07:
HI,
how can I get my epaper 2in7b (rpi HAT version with 4 buttons) working?
demo from waveshare was running.

on local stratux:
installed the SW according your docu.
I created a new dir: /home/pi/stratux-radar-display/main/displays/Epaper_2in7b
copied all contents (also subdirs) from /home/pi/stratux-radar-display/main/displays/Epaper_1in54
copied epd2in7b.py  epd2in7b_V2.py from waveshare

    # Pin definition
    RST_PIN         = 17
    DC_PIN          = 25
    CS_PIN          = 8
    BUSY_PIN        = 24

changed in file controller.py
line 34: from . import epd2in7b
line 150: device = epd2in7b.EPD()

in stratux_radar.sh:
cd /home/pi/stratux-radar-display/main && python3 radar.py   -d Epaper_2in7b &

but nothing works
any ideas?

rudiratlos, 2022-07-07:
ok thanx,
the 3.7" is to big for my case.
What is the SKU (OrderNumber from waveshare for the 1.54")?
SKU 12955?
[https://www.waveshare.com/1.54inch-e-paper-module.htm?amazon](url)

TomBric, 2022-07-08:
Concerning 1.54 there are two options:
Waveshare 12955 1.54inch epaper Display (hat is included) or Waveshare 12561+hat 13512.
I recommend 12955.

Thomas

closed 2022-07-20 by TomBric