TomBric/stratux-radar-display #59: Wrong SSID in instructions for OLE image

wadoadi, 2023-04-01:
In the opening page it gives the connection details as SSID Stratux-Europe, IP 192.168.10.1. however, the connection wasn't made using that SSID. the required SSID for the default connection using the OLE image was stratux  (all lowercase).

Now to add the CO. This is really cool thank you for your work.

TomBric, 2023-04-02:
I do not know whether I got your comment right. Did you mean in the readme on the github  web page?
I modified the position of the quotation mark. Is that what you commented?

wadoadi, 2023-04-02:
hi Tom, its probable just how I read the Read.me "Remark: Current configuration is for **Stratux-Europe** on IP address 192.168.10.1. If you have a different configuration please update /home/pi/stratux-radar-display/image/stratux_radar.sh accordingly."

I took the Stratux-Europa before the IP to be the SSID
to make Idot proof maybe add SSID = stratux

TomBric, 2023-04-03:
Ok, understood and done.

Thomas

closed 2023-04-03 by TomBric