TomBric/stratux-radar-display #61: Not Working with Stratux Europe Edition 1.6r1-eu029 ?

gauthier108, 2023-09-04:
Hello everyone,

I just reflashed my stratux sd card with the new stratux europe version (eu029).
Then I've re installed stratux-radar-display following the instruction for installation on an already existing stratux, without any errors.
But nothing is happening on the OLED screen ...

When i flash my old image again, everything is okay.

Do someone have an idea?

TomBric, 2023-09-04:
Hi gauthier108,

stratux-radar-display is working with eu029. I am using both successfully together.

Can you please try the new development version of stratux-radar-display (ver. 1.9). 
To do this, please use this (as step 4):

git clone -b dev https://github.com/TomBric/stratux-radar-display.git

This should work. 
I will probably soon make this version available as the production one.

Thomas

gauthier108, 2023-09-04:
Thank you very much for your fast answer!
Just tried it, and it's working !!

Many thanks

TomBric, 2023-09-04:
Fine. 
I will make the dev version official soon after some more tests.

Thomas

closed 2023-09-04 by TomBric