TomBric/stratux-radar-display #63: somthing funny with the install script

Puppa-Smurf, 2023-12-24:
I have followed the install instructions onto afresh  Pi zero W 

git clone https://github.com/TomBric/stratux-radar-display.git  <--Was ok 
as user pi. "/bin/bash /home/pi/stratux-radar-display/image/configure_radar.sh <- there was an error in that it complains that luma-core 2.4.1 requires pillow>=9.2.0, but you have pillow 8.4.0 which is incompatible. the funny thing is you can see that it then Found existing installation: Pillow 10.1.0, but removed it and insurted Successfully installed pillow-8.4.0! then whinges that it cant use it.

Any Idears?

Collecting pillow==8.4
  Using cached https://www.piwheels.org/simple/pillow/Pillow-8.4.0-cp39-cp39-linux_armv6l.whl (1.1 MB)
Installing collected packages: pillow
  Attempting uninstall: pillow
    Found existing installation: Pillow 10.1.0
    Uninstalling Pillow-10.1.0:
      **Successfully uninstalled Pillow-10.1.0**
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
**luma-core 2.4.1 requires pillow>=9.2.0, but you have pillow 8.4.0 which is incompatible.**
Successfully installed pillow-8.4.0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'piwiz' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
fatal: destination path 'stratux-radar-display' already exists and is not an empty directory.
sed: can't read /etc/pulse/system.pa: No such file or directory

TomBric, 2023-12-24:
Hi,
unfortunately pillow changed some functions, so that the radar is not fully compatible with the new version.
I already changed as necessary for the epaper display, but not yet for the OLED display.

So I just updated the install script, which now installs pillow version 9.5 which should work. If you remove your old installation (e.g. by "mv stratux-radar-display old.stratux-radar-display" and the again "clone .." this error should be away.

Concerning the second topic, I am not clear yet, why this happens. I will check. Probably some inconsistency with the new raspian versions.

Puppa-Smurf, 2023-12-25:
Thanks, and Happy Christmass.

Looks like your scrip changed worked. Now I just have to work out the next bit ;) 

******
Installing collected packages: pillow
  Attempting uninstall: pillow
    Found existing installation: Pillow 10.1.0
    Uninstalling Pillow-10.1.0:
      Successfully uninstalled Pillow-10.1.0
**Successfully installed pillow-9.5.0**
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'piwiz' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
fatal: destination path 'stratux-radar-display' already exists and is not an empty directory.
sed: can't read /etc/pulse/system.pa: No such file or directory
sed: can't read /etc/pulse/system.pa: No such file or directory
sed: can't read /etc/pulse/system.pa: No such file or directory
addgroup: The user `pulse' does not exist.
addgroup: The group `pulse-access' does not exist.
Radar configuration finished
*****
**Stratux2 is the Radar Screen P1**
pi@Stratux2:~/stratux-radar-display/image $ ping 192.168.20.47
PING 192.168.20.47 (192.168.20.47) 56(84) bytes of data.
64 bytes from 192.168.20.47: icmp_seq=1 ttl=64 time=9.04 ms
64 bytes from 192.168.20.47: icmp_seq=2 ttl=64 time=3.76 ms
64 bytes from 192.168.20.47: icmp_seq=3 ttl=64 time=5.52 ms
64 bytes from 192.168.20.47: icmp_seq=4 ttl=64 time=7.13 ms
64 bytes from 192.168.20.47: icmp_seq=5 ttl=64 time=6.24 ms
**192.168.20.47 is the stratux server**

**Trying to run the script** 
pi@Stratux2:/ $ cd /home/pi/stratux-radar-display/main && python3 radar.py -b -d Oled_1in5 -c 192.168.20.47 &
[1] 9180
pi@Stratux2:/ $ Traceback (most recent call last):
  File "/home/pi/stratux-radar-display/main/radar.py", line 43, in <module>
    import radarbluez
  File "/home/pi/stratux-radar-display/main/radarbluez.py", line 36, in <module>
    import pydbus
  File "/usr/local/lib/python3.9/dist-packages/pydbus/__init__.py", line 1, in <module>
    from .bus import SystemBus, SessionBus, connect
  File "/usr/local/lib/python3.9/dist-packages/pydbus/bus.py", line 1, in <module>
    from gi.repository import Gio
ModuleNotFoundError: No module named 'gi'
^C
[1]+  Exit 1                  cd /home/pi/stratux-radar-display/main && python3 radar.py -b -d Oled_1in5 -c 192.168.20.47
pi@Stratux2:/ $

*******
I then realised that the gi must be missing! 
run sudo apt install python3-gi
then run the script again, and the screen sprang into life. ;) 

Thank you and happy Christmas!

TomBric, 2024-02-18:
Included in version 2.0

closed 2024-02-18 by TomBric