TomBric/stratux-radar-display #72: 3.7" E-Ink display not working at all

robtek57, 2024-06-15:
Hi,
I've gotten an Pi zero 2W , the 3.7" E-Ink display with the Driver Hat, programmed the Pi with the 2.03 Version and the display didn't do anything.
Then I programmed a Pi4 with the Stratux 030 Version you adapted for the E-Ink display, that didn't boot.
Both Pi's worked with the official Software, could you help me troubleshooting?

TomBric, 2024-06-16:
Hi,
first important check is to check the cable and the connections on the GPIO auf the Pi. Are they all in the right order? 
Which driver hat version do you have? If it is 2.3 please make sure to also connect PWR on pin 1.
Second check would be to use an older version of the radar-software. Please try version 1.9 (https://github.com/TomBric/stratux-radar-display/releases/tag/v1.9). I had to change a lot for bookworm in Version 2.0.
Check the switches on the hat: make sure to switch the "Display Config" switch to A. "Interface Config" is to be switched to 0 (4-line SPI).
If you still have nothing on the display, do the following:  With a PC connect to the stratux network. Login on stratux by "ssh pi@192.168.10.1".  If you get a prompt get the IP of your display with "arp -a". You should see several IPs, one of  this is your PC, the other the Pi of the display (probable 192.168.10.35 or 42). From the stratux logon to the radar-display by "ssh pi@192.168.10.xx" with the IP-addr of your display. If you get a logon on the display, you can do "cd stratux-radar-display/main" and manually start "python3 radar.py -d Epaper_3in7 -v 1". Then you get a debug output, which can help us to sort out the problem.

robtek57, 2024-06-16:
Thanks for the answer, will do

If you still have nothing on the display, do the following: With a PC connect to the stratux network. Login on stratux by "ssh ***@***.***". If you get a prompt get the IP of your display with "arp -a". You should see several IPs, one of this is your PC, the other the Pi of the display (probable 192.168.10.35 or 42). From the stratux logon to the radar-display by "ssh ***@***.******@***.***>" with the IP-addr of your display. If you get a logon on the display, you can do "cd stratux-radar-display/main" and manually start "python3 radar.py -d Epaper_3in7 -v 1". Then you get a debug output, which can help us to sort out the problem.

robtek57, 2024-06-18:
Tried the 1.9 Version, got 2 IP's with arp -a from stratux, one with the ending 18 and the other with 42. Both give me a connection refused.

TomBric, 2024-06-20:
Good morning,
did you activate ssh during writing the image to the SD?
Please use Pi Imager when writing the image with the following options:
- set a hostname (e.g. radar.local)
- set a username and password (e.g. "pi" and "raspberry" or whatever you can remember)
- set Wifi: "stratux" and leave the password empty
- set language options to your preferences
- activate ssh, if you want to login later 

[image]
[image]

robtek57, 2024-06-20:
Did everything as described, no joy.
I did find out that there is apparently no connection to the stratux, because when I manually start radar.py I get a endless retry to reconnect from the situationhandler.

TomBric, 2024-06-23:
Strange. The display should come up also without a connection to stratux.
Remotely I actually do not know what to recommend additionally. There seems to be a problem with the display. 
If you can send pictures of your cable connections and the log when starting manually, maybe I get another idea.

robtek57, 2024-06-23:
Thanks for the info, will do in the next days