TomBric/stratux-radar-display #74: 1.5" oled display issue

dbr259, 2024-07-19:
Hello Tom, following the instructions you posted, the OLED screen should be installed on the installed stratux, but it does not do so because there are errors from the repositories and the library. My question is if you can prepare and put it on github. stratux-eu31-with-Oled_1in5-v2.03-d56ca413.img.zip

TomBric, 2024-07-21:
Done. Unfortunately I do not have an Oled available right now, so I could not check.
Give me a feedback, whether it works.

dbr259, 2024-07-21:
Strux works, unfortunately not OLED
[image]
[image]

TomBric, 2024-07-21:
Thanks for the offer. I have got enough of the OLEDs available. I just need to built a test system.
I will do that tomorrow.

TomBric, 2024-07-22:
I uploaded a new version for the Oled in 2.03 release. Its working in my test. Please test yourself as well.
There was an "apt update" missing in the install script.

dbr259, 2024-07-22:
Its working ..thanks Tom.

closed 2024-07-22 by TomBric