TomBric/stratux-radar-display #76: Stratux-radar-display not working :-(

gauthier108, 2024-07-29:
Hello everyone !

Sorry to bother you, but I updated my stratux from scratch. I have in fact installed version 1.6r1-eu030 of the Stratux and 2.0.3 of the radar display.
However, when powering on, the screen often does not start. In debug mode I get the following error message.
Would any of you know how to help me?

[image]

TomBric, 2024-07-29:
Hi, 
there is a python library missing. 
   pip3 install  ADS1x15-ADC --break-system-packages
should correct it. 
How did you install? Did you use a ready made image or install yourself?
If you installed yourself did you clone the repo locally or use the zip or tar files?

gauthier108, 2024-08-06:
Hi Tom,
I installed it myself on a existing stratux (last eu version).
I think i followed the expert setup and cloned the repository.
What is strange is that the stratux-radar-display is sometimes working (one start in 5 is ok).
I will try to manually install the library and will keep you in touch.

Many thanks

closed 2024-11-28 by TomBric