TomBric/stratux-radar-display #77: Stratux SSID, Bluetooth issues

TG-35, 2024-08-06:
Hi,

First, thank you very much for your great job!
I recently built a stratux display ( version 2.03 with 1,54 inc e-paper display) to get Traffic information.
I noticed 2 problems:
1- the display is working only with the by default stratux Wifi SSID (=‘stratux’). I tried to change the wifi SSID matching with my stratux (‘stratux XX’) using PI Imager or the stratux display configuration menu (requesting a reboot after the change) without success. Is it possible to connect the display to a stratux with an SSID like ‘stratux XX’?
2 - Bluetooth: I managed to connect my Bose A20 once (after the scan/pairing phase), but after stopping and restarting the equipment, the display no longer reconnects to the Bose A20…

Last point, as I do not use the G-meter, AHRS and preflight checklist modes, would it be possible to add the functionality to deactivate then (eg in the configuration file)

regards

TomBric, 2024-08-06:
Hi TG-35,
1. SSID:  yes, that is possible. I also use this regularly. It seems like you are using a blank in the SSID. This is not allowed. Please use "-" or "_" instead.
2. Can you check starting "bluetoothctl" in an command window? Then you can retrieve some information on the bluetooth status. Bluetooth is really tricky and it may hang somewhere.

Concerning your last point:  Yes, this is already implemented. You can use the "-modes" option in image/stratux_radar.sh.
So the line then could be:
`cd /home/pi/stratux-radar-display/main && python3 radar.py -r -d Epaper3in7 -modes RADCM -c 192.168.10.1`

Here is the list of modes available:

-modes DISPLAYMODES, --displaymodes DISPLAYMODES
                        Select display modes that you want to see R=radar T=timer A=ahrs D=display-status
                        G=g-meter K=compass V=vsi I=flighttime S=stratux-status C=co-sensor M=distance
                        measurement L=checklist Example: -modes RADCM

Thomas

TG-35, 2024-08-06:
Hello Thomas,Thanks for your quick answer.1-SSID: I may be doing something wrong! I tried an SSID without space as suggested, with PI Imager and in the display menu, but it does not work. Even if the display is saying « rebooting… » it is not the case. And when I force the reboot by cutting the power, it restart with SSID « stratus ».Not so important, I have changed again the stratux SSID to « stratux » on my Stratux, so it works.2- Bluetoothctl: I did not succeed up to now to get an SSH connection to the PI zero 2 W (ssh is enabled with PI Imager). No ethernet connexion on the zero PI (I tried an USB to ethernet adaptor without success),  no way with USB on my windows PC. May be OK when both PC and display are connected to the stratux?3-Displaymodes:modification done, thanks!Regards,Thierry

Hi TG-35,

TomBric, 2024-08-06:
Hi Thierry, for connecting via ssh to the display, you have to use the stratux wlan with your pc. Then have a look at the stratux log file (->logs -> stratux.log). There you can look for "radar". Then you can identify the IP address of the radar display (e.g. 192.168.10.24). Then connect via ssh: ssh pi@192.168,10.xxxx
Thomas

TomBric, 2024-08-06:
Hi Thierry, I just rechecked setting the stratux ssid differently. It works for me. Did you set the Wifi country correct in stratux and as well in the Pi Imager?

TG-35, 2024-08-29:
Hi Thomas,
I have done a new try with SSID « stratuxXX » and it works well now. Thank you

TG-35, 2024-08-29:
Windows PowerShell
Copyright (C) Microsoft Corporation. Tous droits réservés.

Testez le nouveau système multiplateforme PowerShell https://aka.ms/pscore6

PS C:\Users\TG> ssh ***@***.***
The authenticity of host '192.168.10.36 (192.168.10.36)' can't be established.
ECDSA key fingerprint is SHA256:fspqK409W+En3ccpkApA5yyayvln1cqwYbDXfHt0ptY.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.10.36' (ECDSA) to the list of known hosts.
***@***.***'s password:
Linux radar 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

***@***.***:~ $ sudo -s
***@***.***:/home/pi# bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:B4:42:07 Pairable: yes
[Bose A20]#

PS C:\Users\TG>
PS C:\Users\TG> ssh ***@***.***
***@***.***'s password:
Linux radar 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Aug 28 19:08:21 2024 from 192.168.10.42

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

***@***.***:~ $ sudo -s
***@***.***:/home/pi# bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:B4:42:07 Pairable: yes
[CHG] Device 28:11:A5:91:C9:38 Connected: no
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:B4:42:07 Discovering: yes
[NEW] Device 4C:52:50:BC:5F:BD 4C-52-50-BC-5F-BD
[NEW] Device 46:AD:ED:2F:AD:75 46-AD-ED-2F-AD-75
[NEW] Device 58:CD:EF:E8:16:6B 58-CD-EF-E8-16-6B
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -55
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -43
[NEW] Device 51:48:22:91:B3:09 Steel HR B4
[CHG] Device 28:11:A5:91:C9:38 Connected: yes
[CHG] Device 28:11:A5:91:C9:38 Connected: no
[CHG] Device 28:11:A5:91:C9:38 RSSI: -43
[CHG] Device 28:11:A5:91:C9:38 TxPower: 4
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -43
[DEL] Device 46:AD:ED:2F:AD:75 46-AD-ED-2F-AD-75
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -33
[CHG] Device 51:48:22:91:B3:09 RSSI: -92
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.InProgress
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -55
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -54
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -43
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -34
[bluetooth]# scan off
Discovery stopped
[CHG] Device 28:11:A5:91:C9:38 TxPower is nil
[CHG] Device 28:11:A5:91:C9:38 RSSI is nil
[CHG] Device 51:48:22:91:B3:09 TxPower is nil
[CHG] Device 51:48:22:91:B3:09 RSSI is nil
[CHG] Device 58:CD:EF:E8:16:6B RSSI is nil
[CHG] Device 4C:52:50:BC:5F:BD TxPower is nil
[CHG] Device 4C:52:50:BC:5F:BD RSSI is nil
[CHG] Controller B8:27:EB:B4:42:07 Discovering: no
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:B4:42:07 Discovering: yes
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -55
[CHG] Device 4C:52:50:BC:5F:BD TxPower: 11
[CHG] Device 51:48:22:91:B3:09 RSSI: -80
[CHG] Device 51:48:22:91:B3:09 TxPower: 0
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -43
[CHG] Device 28:11:A5:91:C9:38 RSSI: -52
[CHG] Device 28:11:A5:91:C9:38 TxPower: 4
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -38
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -33
[CHG] Device 28:11:A5:91:C9:38 RSSI: -42
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -55
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -49
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -43
[CHG] Device 51:48:22:91:B3:09 RSSI: -67
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -54
[NEW] Device 6D:20:80:EE:51:7F 6D-20-80-EE-51-7F
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -37
[DEL] Device 51:48:22:91:B3:09 Steel HR B4
[CHG] Device 6D:20:80:EE:51:7F RSSI: -93
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -34
[CHG] Device 58:CD:EF:E8:16:6B RSSI: -48
[CHG] Device 4C:52:50:BC:5F:BD RSSI: -59
[bluetooth]# scan off
[CHG] Device 6D:20:80:EE:51:7F RSSI is nil
[CHG] Device 28:11:A5:91:C9:38 TxPower is nil
[CHG] Device 28:11:A5:91:C9:38 RSSI is nil
[CHG] Device 58:CD:EF:E8:16:6B RSSI is nil
[CHG] Device 4C:52:50:BC:5F:BD TxPower is nil
[CHG] Device 4C:52:50:BC:5F:BD RSSI is nil
[CHG] Controller B8:27:EB:B4:42:07 Discovering: no
Discovery stopped
[DEL] Device 6D:20:80:EE:51:7F 6D-20-80-EE-51-7F
[bluetooth]#

closed 2024-08-30 by TomBric