TomBric/stratux-radar-display #81: CO sensor causes problems

DocTopi, 2024-10-08:
Hi TomBric,
thanks a lot for your great development.

I´ve installed the stratux-radar-display v2.03 on a Zero 2 W with the Oled Display 1.5 inch and Benewake TFMini Plus Lidar Sensor.
Host is Stratux Europe Edition 1.6r1-eu032.

During testing on the table everything is working well, but after installing MICS-5524 sensor with converter ADS1115 the radar-display with radar.py parameter -z starts completely, the left and the right button are working correctly, but after pushing the middle button the radar-display freezes at once.
After starting with radar.py parameter - w the radar-display freezes with the initial startscreen,
after starting with radar.py parameter - nc everything is working well again.

Are Oled Display 1.5 inch and MICS-5524 sensor with converter ADS1115 operational together?

Are there possibilities for testing the sensor and the converter?

A lot of thans for your support.
Kindly Franz

TomBric, 2024-10-13:
Hi Franz,
I found the error and corrected it in the "dev" branch.
I actually do not have a test device with CO and Oled at hand. Can you do a test by cloning the "dev" branch. 
Otherwise I can also prepare a new image for you.
Thomas

DocTopi, 2024-10-13:
Hi Thomas,
Sorry, I`m a Linux NewBie,
Please prepare an image for the Oled-Display with Zero 2 W and I`ll test it at once.
The image for the 1.54 Epaper Display seems to have the same problem: a completely new hardware installation freezes, but with argument "-nc" everything is working well ...
Please prepare an image for the Epaper Display too.
Thank you very much !!!
 Franz

TomBric, 2024-10-14:
Hi Franz,
I just uploaded version 2.05.
Please check. 
Thomas

DocTopi, 2024-10-14:
Good evening Thomas,
great, the v2.05 images for the 1.54 Epaper and the OLED Display are working well ...
Thanks a lot !!!
Franz
[image]

closed 2024-10-18 by TomBric