TomBric/stratux-radar-display #87: Radar showing "noise" instead of radar

Stefan-Kosker, 2024-11-25:
I installed the stratux EU32 and radar software afterwards. Today I installed the SDR devices and can see traffic ADS-B in Stratux.

[image]

But the screen shows this: 
[image]

What did I wrong? I am using the same screen you suggested and checked the connections again and again.

TomBric, 2024-11-26:
Cabling seems correct.
Did you **disable** I2C/OGN Transmission under settings?  (see required setting in picture)

[image]

Stefan-Kosker, 2024-11-26:
[image]

Ye. Disabled. Still problem. 

Does it have to be disabled at installation time?

TomBric, 2024-11-26:
It is ok to disable it and then reboot, not necessary at installation time.
And this does NOT affect T-Beam. It is only needed for a special hardware, a GPIO hat. So no need for an additional pi.

But still your problem exists. I will check with my hardware in the next days. Strange ...

Stefan-Kosker, 2024-11-26:
Appreciate it. I would really like to not having to buy another end device. Because my next step will be classic over engineered German solution with a mini pc, hdmi touch screen and shit. Then I can throw away the whole energy planning with 30Ah powerbank out of window...

Thank you mate.

Stefan-Kosker, 2024-11-28:
Ye Display works.

TomBric, 2024-11-29:
Hi Stefan, I just checked with my hardware and the oled display:
- Installed stratux 032 on a raspberry pi 3B
- stratux connected to my home network and persistent logging on
- started and stopped timesync to get proper time (gps was not yet available)
- sudo apt update  (install git did not work the first time)
- sudo apt install git
- git clone https://github.com/TomBric/stratux-radar-display.git
- sudo /bin/bash /home/pi/stratux-radar-display/image/configure_radar_on_stratux.sh
- checked image/stratux_radar.sh: oled was already configured
- rebooted
- Radar 2.05 is working
Can you please retry a full installation? Maybe you received outdated libraries due to missing proper time.
Thomas

Stefan-Kosker, 2024-11-29:
Checklist:

- Flashed stratux-v1.6r1-eu032-ff1f01dc.img into a sandisk32gb microSSD
- Stratux connected to my home network and tested for functionality:
[image]

- persistent logging on, OGN Transmission via I2c off:
[image]

- started and stopped timesync to get proper time (gps was not yet available)
[image]

- sudo apt update (as suggested in the PR)
[image]

- sudo apt install git: Done
- git clone https://github.com/TomBric/stratux-radar-display.git
[image]

- sudo /bin/bash /home/pi/stratux-radar-display/image/configure_radar_on_stratux.sh
[image]

- checked image/stratux_radar.sh: oled was already configured
[image]

- rebooted
Power off and on

Result: Still failing.

I will send raspberry and the screen back and purchase a new one. Maybe it is a hardware fail. Theoretically, when I push SD Card into the new Raspberry, it should work out without any issues.

TomBric, 2024-11-29:
I am still considering a cabling or software problem.
Before you send everything back, maybe you can do a test with luma, which is the oled software.
See examples and instructions here:
github.com/rm-hull/luma.examples
This way you can test display and pi together independent of stratux.

Stefan-Kosker, 2024-11-29:
Yeah no, so the luma.examples I couldn't run. Waveshare test just stays black. Sad

TomBric, 2024-11-30:
I am actually using a pi 3B v1.2.
Same as yours.
But also with a Pi4 it works.

Stefan-Kosker, 2024-11-30:
Allright. Then lets do it easier. Is a HAT possible? I already ordered a new 3b+ and would also order this one:
https://amzn.eu/d/hlwbcrM

Any experience with HAT 3.7inch? Has 3way and 4way spi, I will not use the buttons. This way I shouldn't have any problems with wrong cabling etc. Just place it on the rasp pi and 3d print a proper box with that.

What do you think?

TomBric, 2024-11-30:
The hat is possible, it works on additional pi zero and also with a pi3b.
The problem with the hat is that you are not able to connect additional devices like baro and ahrs to stratux.
So in your case it is not a proper solution.
BTW: Did you also try the full image I provide with stratux and oled from github, see here
`https://github.com/TomBric/stratux-radar-display/releases/download/v2.05/stratux-eu032-with-Oled_1in5-v2.05-00e2d1ef.img.zip`

Stefan-Kosker, 2024-11-30:
Will try it later, currently I am driving for a nice flight :)

Since I am going to use t beam, baro is going to be on Io of tbeam i2c. According to viruspilot his docs, this should work.

TomBric, 2024-11-30:
Baro on T-Beam is possible, but AHRS not.
Have fun on your flight. I also wanted to fly this morning, but Koblenz was still foggy.

Stefan-Kosker, 2024-12-01:
> The hat is possible, it works on additional pi zero and also with a pi3b. The problem with the hat is that you are not able to connect additional devices like baro and ahrs to stratux. So in your [...]

[image]

Sad but no chance

TomBric, 2024-12-01:
I just looked at your picture: You are using a different OLED Module!
Software is made for waveshare 14747, which is an RGB module. Somehow your module looks different compared to mine.
Please check your order. It has to be set to 4-SPI.
[https://www.waveshare.com/1.5inch-rgb-oled-module.htm](url)

Stefan-Kosker, 2024-12-04:
E-Paper Hat works flawlessly. Tells me even noGPS. Thank you

TomBric, 2024-12-05:
Good to hear!

closed 2024-12-05 by TomBric