TomBric/stratux-radar-display #91: Where can i find the Display Configuration?

Polzi, 2025-01-29:
Hi There, 
i am using an OLED Display, in Version 2.6 i should be able to select the display in the Webconfig.
But i can not find where.
Thank You for help,
Robert

TomBric, 2025-01-29:
Hi Robert,
thanks for this. You spotted a mistake in the web configuration, when using the software directly on the stratux.
I will provide an update of 2.06 soon.
If you have some expert knowledge you can change the lines in stratux-radar-display/main/radar-web/templates/index.html:

Old (error):
{% if on_stratux == False %}
        {{ render_field(radar_form.stratux_ip, form_type='horizontal') }}
        {{ render_field(radar_form.display, form_type='horizontal') }}
{% endif %}

New:
{% if on_stratux == False %}
        {{ render_field(radar_form.stratux_ip, form_type='horizontal') }}
{% endif %}
{{ render_field(radar_form.display, form_type='horizontal') }}

Thomas

Polzi, 2025-01-29:
Hi There,
so it´s not that i am blind... ;-)
unfortunately my programming skills are not very good.
I will wait for the next release, i have enough to do with ma redesign of the case.
Thank You,
Robert

TomBric, 2025-02-01:
Hi Robert,
a new image with the bug fix is on github. You can find it here:
https://github.com/TomBric/stratux-radar-display/releases/download/v2.06/stratux-eu32-radar-webconfig-v2.06-1c424640.img.zip

Thomas

Polzi, 2025-02-01:
Thank You, 
works perfect now!

Robert

closed 2025-02-02 by TomBric