TomBric/stratux-radar-display #93: Ghost and full refresh epapaer 3.7

vkeven, 2025-04-27:
Is there an automatic way to refresh the screen from time to time , there is a lot of faded view and and refreshing the screen manually solved this but its unpractical during flight... , Oled is okay but small...

TomBric, 2025-04-28:
Actually there is no way for an automated refresh. I had it in the beginning, but ghosting was never a real issue for most users, since the displays did not show much ghosting and it is a little bit disturbing in flight, if the display "flickers".
But of course I can implement a feature to have a configurable timer for an auto refresh. 
Is that what you need? E.g. an automatic refresh after a lap of 5 or 10 minutes?

vkeven, 2025-05-02:
YEs it would be nice !

TomBric, 2025-08-05:
Implemented in v2.10
Configurable via webconfig

closed 2025-08-05 by TomBric