TomBric/stratux-radar-display #94: Crash of brcmfmac after radar-display installation

vkeven, 2025-05-20:
After installation on stratux 1.6eu32 there is a crash of wifi after reboot and stratux_radar stopped loading , ieeee80211 , firmware has halted or crashed , currently investigating

TomBric, 2025-05-21:
This is a typical sign, if you are having power supply issues. Please check your power supply and power cables.
You can also check kernel logs (with "dmesg"). Check if there is "undervoltage detected".

vkeven, 2025-05-21:
Look like ads115/cowarner initialization issue

Uncaught exception: ValueError: math domain error
Traceback (most recent call last):
File "/home/pi/stratux-radar-display/main/radar.py", line 943, in <module>
    main()
File "/home/pi/stratux-radar-display/main/radar.py", line 820, in main
    asyncio.run(coroutines())
File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
File "/home/pi/stratux-radar-display/main/radar.py", line 787, in coroutines
    await asyncio.gather(tr_handler, sit_handler, dis_cutoff, u_interface, sensor_reader, ground_sensor_reader)
File "/home/pi/stratux-radar-display/main/cowarner.py", line 327, in read_sensors
    changed = read_co_value()
              ^^^^^^^^^^^^^^^
File "/home/pi/stratux-radar-display/main/cowarner.py", line 201, in read_co_value
    ppm_value = round(ppm(rs_gas / r0))
                      ^^^^^^^^^^^^^^^^
File "/home/pi/stratux-radar-display/main/cowarner.py", line 105, in ppm
    val = 10 ** ((math.log10(rsr0) - 0.9) / -0.75)
                  ^^^^^^^^^^^^^^^^

vkeven, 2025-05-21:
After some test we reload the image _stratux-eu32-radar-webconfig-v2.06-1c424640.img_  ( stratux-radar-display image ) and the stratux network part won't load correctly at 75% of the time after a reboot ( 3b+ , 978mhz ,1090mhz , GPS , GY-91 ) , network crashed but radar-display load correctly with no connection on screen , the symptom is the same either we installed it over stratux or we load the ready image , I did a apt-get upgrade and after new network firmware package instaled during update the  network load correctly at each reboot... , for the CO warn error ist look like my ADS1115 is defective or mics-5524 is defective ,  each time I try to load it radar-display crash at load so I change stratux_radar.sh and add -nc to prevent module load and problem solved , it would be nice than when you choose diable into web interface like co warner the script would change with -nc directly

TomBric, 2025-05-22:
- if you use a co-sensor please install stratux-radar-display on a seperate Pi zero2, not on stratux itself. Due to conflicts with stratux i/o this is the way forward
- for the installation of the radar on a separate Pi there is the option to suppress CO-activation, see here:
[image]
- btw: I also added the option on the web interface which you recommended to do automatic refresh, but it's not yet fully implemented

- there have been some changes in the python base installation. Can you please use the latest release 2.07 for your installation, see here:    https://github.com/TomBric/stratux-radar-display/releases

vkeven, 2025-05-22:
Ok i will try with version 2.0.7 but I can confirm than this worked during my breadboard test with the same setup with complete stratux install , I tried with brand new gy91 ads1115 and mics5524 and sometime the co initialize and sometime it crash , could not pin point exactly where is the culprit... Thanks for the new code ! will tried this

TomBric, 2025-05-23:
I have just checked in the full functionality for a time based automatic full refresh in the dev branch.
It would be nice if you can also test it. You can install the new version 2.08 as desribed here:
https://github.com/TomBric/stratux-radar-display/wiki/Expert-installation-on-a-separate-Pi-Zero2
Just change the git clone command. Instead of 
`git clone https://github.com/TomBric/stratux-radar-display.git`
use the dev branch with -b like this:
`git clone -b dev https://github.com/TomBric/stratux-radar-display.git`
Also the Co-sensor crash should no more happen. But nevertheless your sensor seems to have a problem

closed 2026-01-04 by TomBric